ataradov/free-dap #1: checking if works with atmel studio

ladyada, 2016-09-08:
heya i did a quick build onto an atsamd21, but don't have a target attached. connecting with atmel studio 7 gives:

> Unable to connect to tool EDBG (123456)
> 
> Timestamp:	2016-09-08 00:07:32.092
> Severity:		ERROR
> ComponentId:	20000
> StatusCode:	0
> 
> Failed to set-up tool (no context id returned).

do you know if it definitely does or does not work with AS7?

normally if you dont have a target you can still connect to EDBG so i dont /think/ that's an issue. if it doesn't work as-is with atmel studio 7 thats ok i can hack on making that work. just checking before i pull out the usb analyser and go down that path :smile:  
thx!

ataradov, 2016-09-08:
I have not tried to connect it to AS. I'm surprised it worked to this extent without changing VID/PID.

AS relies on additional vendor-specific commands and here they are not implemented, obviously.

We can try to log what commands are written and at least return OK response, but if they require actual payload in the response, then we'll have to figure out what those commands are.

I'll try to do that today.

ataradov, 2016-09-08:
Although if you have a hardware USB analyzer, it would not hurt to do a scan of a real EDBG chip connecting and working for a few seconds, but not absolutely necessary at this point.

ataradov, 2016-09-08:
I found some of my old USB logs for real EDBG on Xplained Pro board. It looks like there are some vendor-specific commands that identify the kit, they would probably have to return something meaningful.

It would make sense to make new logs from Atmel-ICE as opposed to Xplained kits.

ladyada, 2016-09-08:
k for my ref here's the commands
http://www.atmel.com/webdoc/protocoldocs/ch02s02.html

ataradov, 2016-09-08:
That does not help much, since there is no actual description of the command payload. I know that avrdude had some of that reverse-engineered already, but it is very dirty.

But in any case, here is what I have found so far:
1. It looks like AS tries to recognize generic CMSIS-DAP tools, but fails, possibly dies to simple lack of testing in this area.
2. In order for things o move forward I had to assign Atmel-ICE VID/PID and product descriptor and serial number strings. If any of this is not correct, the tool is recognized as a generic tool and fails with the error you see.
3. Once things from item 2 are "correct", AS detects the tool as Atmel-ICE and spams it with vendor command 0x81. I have not tried to parse the payload, since its description is not readily available.

At this point it would be nice to see if AS could just work with generic CMSIS-DAP tools, since faking VID/PID is not a real solution.

ladyada, 2016-09-08:
ya that list is just handy for if/when i look at the packets i can x-reference what they're supposed to be.

i understand better now i thought this was a generic cmsis-dap that AS7 would match up based on descriptor, not an 'ICE' clone. i agree its probably just VID/PID locked.

i am in contact with atmel (er, microchip) folks who have insight into AS7 and - in fact - they pointed me to this repo because i asked for the mEDBG code and the reply was "cant give that to you, check out free-dap"

so, what i think i'll do is see how hard it is to get AS7 folks to unlock the VID/PID requirements and then circle back here and compare traces from my EDBG to this. i only need it to support 1 chip :)

ataradov, 2016-09-08:
Free-DAP is just a free alternative to ARM CMSIS-DAP, but with those locks in place, ARM CMSIS-DAP won't work either.

I'll try to keep track from the inside as well, since it would be beneficial for all involved parties to have generic CMSIS-DAP adapters supported in AS.

Email may be a better channel for this. Contact me at alex@taradov.com if that works for you.

closed 2021-05-16 by ataradov