ataradov/free-dap #3: What is gmkdir?

ooxi, 2018-02-09:
I'm trying to compile free-dap on Ubuntu 17.10 for samd11. However when trying to `make` [Makefile_std](https://github.com/ataradov/free-dap/blob/master/platform/samd11/make/Makefile_std), I'm getting the following error:

```
$ make -f Makefile_std 
make: gmkdir: Command not found
Makefile_std:64: recipe for target 'directory' failed
make: *** [directory] Error 127
```
Replacing `@gmkdir` with `@mkdir` fixes the problem.

What is `@gmkdir`, why is it used instead of `@mkdir` and how can I provide `@gmkdir` on Ubuntu?

ataradov, 2018-02-09:
So this is very stupid thing. On Windows standard mkdir does not understand '-p' option, and just creates a '-p' directory. But you can't just replace internal mkdir with external one, so the version of shell utils that comes with Atmel Studio provides GNU mkdir that is called gmkdir. 

To make it work on Linux, just remove 'g' from the name, or create an alias gmkdir -> mkdir.

I'd like to find a better solution, bit I'm not sure there is one.

ooxi, 2018-02-09:
That's ok, I'll just make an alias from `gmkdir` to `mkdir` on my system :-)

closed 2018-02-09 by ooxi