ataradov/free-dap #4: Trouble when free-dap hardware is connected to target but not to power

ooxi, 2018-03-11:
Hi,

I've been running into trouble when connecting my free-dap hardware (based upon d11_usb_mini) with the target but not powering the free-dap hardware.

The symptom looks like the target does not work since the target's `nRST` voltage drops from 3,3 V to 1,3 V. Since the target tries to hold 3,3 V on `nRST`, the Atmel SAMD 11 on the free-dap hardware is subject to a voltage between PA09 and GND (connected via J7/J8 to the target).

Together with a pal we came up with the following circuit:

[image]

This should disconnect PA09 from J7 unless `V_USB` is present. Of cause, the logic will be inverted (PA09 becoming RST instead of ¬RST) but that can be fixed in software fairly easy. When `V_USB` is not present, the pulldown will make sure that T2 is disconnected (GND is present since it's connected to the target via J8).

I have a couple of questions:

1. Have you experienced this behaviour, too (target not working when connected to unpowered free-dap hardware)?
2. Does this circuit fix my the behaviour or have we overlooked something else?

-- ooxi

ataradov, 2018-03-11:
1. I have never tried, and it was never a design goal to have that scenario working, but I see why it would not work.
2. This should fix the reset, but the other pins will have similar behavior. If you want a complete fix, you need a real bidirectional buffer. But generally there is no limit to how much you can improve the hardware :)

ooxi, 2018-03-11:
Thanks for your feedback! Looking at bidirectional buffer ICs, they are really expensive :D but you are right they seem like the correct way to go.

closed 2018-03-11 by ooxi