ataradov/free-dap #5: Porting to samd21e17

AloyseTech, 2018-08-21:
Hi Alex,

I'm trying to port free-dap on the [Tau](http://rabidprototypes.com/wp-content/uploads/2016/10/Tau-037-schematic.png). I would like to keep the bootloader installed. I've modified the linker script you provide to take care not override the bootloader region and matched the memory map as follow :
`MEMORY
{
  flash (rx) : ORIGIN = 0x00000000+0x2000, LENGTH = 0x20000-0x2000 /* 128k-bootloader */
  ram  (rwx) : ORIGIN = 0x20000000, LENGTH = 0x00004000-4 /* 16k */
}

__top_flash = 0x40000;
__top_ram = 0x20000000 + 0x4000;

ENTRY(irq_handler_reset)
`
I modified the dap_config.h header to match the pinout as follow :
`HAL_GPIO_PIN(SWCLK_TCK,    A, 19)
HAL_GPIO_PIN(SWDIO_TMS,    A, 18)
HAL_GPIO_PIN(TDI,          A, 17)
HAL_GPIO_PIN(TDO,          A, 16)
HAL_GPIO_PIN(nTRST,        A, 7)
HAL_GPIO_PIN(nRESET,       A, 6)

HAL_GPIO_PIN(LED,          A, 27)
`

And the makefile as follow :
`...
LDFLAGS += -Wl,--script=../linker/samd21e17_boot.ld
...
DEFINES += \
  -D__SAMD21E17A__ \
  -DDONT_USE_CMSIS_INIT \
  -DF_CPU=48000000 
...`

Build is succesfull and I flash using the Tau Arduino package Bossac :
`bossac.exe -e -w -v -b free_dap.bin`

Which seems succesfull too:
`Device found on COM15
Atmel SMART device 0x1001000b found
Erase flash
done in 0.405 seconds

Write 7568 bytes to flash (119 pages)
[==============================] 100% (119/119 pages)
done in 0.107 seconds

Verify 7568 bytes of flash with checksum.
Verify successful
done in 0.017 seconds
Set boot flash true
Ignoring set boot from flash flag.`

But when I reset the board, the USB peripheral does not enumerate...
I'm able to re-enter bootloader using the double-tap magic (double press on reset in less than 500ms) from the Tau's bootloader.

I made blink code from the main which works as expected :
`int main(void)
{
  sys_init();
  //dap_init();
  //usb_init();

  HAL_GPIO_LED_out();

  uint16_t tempo = 0xFFFF;
  while (1)
  {
    HAL_GPIO_LED_toggle();
    tempo = 0xFFFF;
    while(tempo--);
  };

  return 0;
}`

Do you know if I'm missing something to make it work on this platform? Could the USB code from the bootloader somehow screw free-dap initialisation?

Thanks for your help :)

ataradov, 2018-08-21:
You probably need to set SCB->VTOR in the startup code. Look at the D11 startup and copy than one line from there.

AloyseTech, 2018-08-21:
It seems to boot correclty now. But I can't use the board using openocd... Do you have an example on how to use free-dap with openocd?
I tried the following command : 
`openocd -f freedap.cfg`
freedap.cfg : 
```
interface cmsis-dap
cmsis_dap_vid_pid 0x6666 0x6666
```

ataradov, 2018-08-21:
No idea, I have not used OpenODC much. This looks like a reasonable tutorial https://mcuoneclipse.com/2015/03/22/openocdcmsis-dap-debugging-with-eclipse-and-without-an-ide/

AloyseTech, 2018-08-21:
Thanks for the link. Could you please share your test sequence? Like flashing an MCU? What tool did you test?

ataradov, 2018-08-21:
I've tested it with IAR, Keil and my own tool https://github.com/ataradov/edbg 
All 3 were working fine.

AloyseTech, 2018-08-21:
I tried using edbg on ubuntu (from windows integration) but it does not find the debugger. I do have the HID Peripheral showing up on windows device manager with VID/PID/SN matching 0x6666/0x6666/123456.

I tried with the following command : 
`./edbg -t atmel_cm0p -s 123456 -r`

But got the following error : 
`Error: unable to find a debugger with a specified serial number`

ataradov, 2018-08-21:
What does "./edbg -l" print?

Why not use native windows version?

AloyseTech, 2018-08-21:
"./edbg -l" show no debugger connected...
I will try the windows version.

ooxi, 2018-09-26:
Sorry for replying to late. Does the Windows Device Manager show the attached device at all? If so, could you please post a screenshot showing the VID/UID? Just want to make sure that USB itself is actually initialized correctly by the MCU.

AloyseTech, 2018-09-26:
This issue was in EDBG itself. It is now solved with latest EDBG version ([see this commit](https://github.com/ataradov/edbg/commit/785f788978c79379a246e98e1760bbdfb340e064)

closed 2018-09-26 by AloyseTech