ataradov/free-dap #8: Suggestion for new board

sslupsky, 2019-05-04:
I have one suggestion for the new hardware board I noticed today.  Would it be possible to provide a jumper so that it is possible to connect nRESET to Pin 3?  Doing so would allow the use of a 1.27mm 6 pin header on the target.

ataradov, 2019-05-04:
I'm not going to be selling those boards, at least not until I'm sure that VCP part is robust and I know I like how the combined device works. I just made a few for testing.

The primary goal of this design was to be used on my development/breakout boards for other device (like EDBG on Atmel kits). This one is the first to use it https://github.com/ataradov/breakout-boards/tree/master/stm32g071c and I like it so far. 

That being said, all design files are open, feel free to modify them however you like.

sslupsky, 2020-08-05:
I have been doing some development with Zephyr and I now appreciate the usefulness of the VCP.  Any chance you have changed your mind and will make these boards available?  Also, have you attempted to run DAPLink on this board?  Is there some things FreeDAP does that DAPLink does not do?

ataradov, 2020-08-06:
I don't really want to make them available for sale generally, but if you order one from Tindie and specify in the order comments that you want a VCP version, I can send you one.

I have not looked at DAPLink, I see no reason to do so. The whole point of this project is to get rid of bloat. Why would I port the bloat here? DAPLink may be better in some aspects, I don't know. But I also don't care, this project serves my needs.

JakeSays, 2021-02-26:
@ataradov DAPLink was the motivation to look at free-dap. I need a dap debugger running on an ATSAME5x. Just spent an hour scratching my head at the DAPLink project. The question I kept asking was "Now exactly why do I need python?". Yeah bloat is right!

closed 2021-05-16 by ataradov