ataradov/free-dap #9: DAP on SAMD51

MonteShaffer, 2019-08-07:
Will this CMSIS-DAP programmer/debugger work on the D51 (M4) or just the D11/D21 (M0+) MCUS?

We have been able to successfully flash D11/D21 with your codebase (using Keil).

In Keil, the DAP debugger is not recognizing the D51 (custom dev board).

Thoughts?

ataradov, 2019-08-07:
It should be fully device-independent. 

Can you try that board with my edbg tool and provide its output? There may be an issue with the board.

MonteShaffer, 2019-08-07:
Based on a previous "issue", I thought maybe the struct SCB->VTOR may be an issue.

We anticipated it to be device-independent.

We will try the EDGB tool and get back to you.

The boards are custom (one with the 64-pin, one with the 128-pin for D51).  So that may indeed be the issue.

We similarly have custom boards for the D11/D21 and they seem to work as expected.

Our boards:  Sasha (D11), Sanja (D21/D51), and Shurik (D51 128 pin).

We are assuming the power coupling are the same for the D21/D51 in the board design.

Chapter 7.2, Figure 7-1 for the D21 [Atmel-42181-SAM-D21_Datasheet.pdf] ... Power Supply Connection
Chapter 7.2, Figure 7-2 for the D51 [60001507A-1130176.pdf] ... Power Supply Connection for Linear Mode Only

Maybe that assumption was wrong!

Anyway, we want to have a project similar to your free-dap flasher that can live on one of the Sasha (D11) boards to flash the other boards, using just the four need pins from the JTAG 10-pins:  SWDIO/SWCLK/RST/GND

We will dig into the EDGB output and get back to you.

Thanks!

MonteShaffer, 2019-08-07:
```
>edgb.exe -b -t samd11 -pv -f IDK.bin
Debugger: Alex Taradov Generic CMSIS-DAP Adapter 8E078C62 v0.1 (S)
Clock frequency: 16.0 MHz
Target: SAM D11D14AM (Rev B)
Programming............................................ done.
Verification............................................ done.
```

This is the SAMD51

```
>edgb.exe -b -t samd51 -pv -f IDK.bin
Debugger: Alex Taradov Generic CMSIS-DAP Adapter 0123456 v0.1 (S)
Clock frequency: 16.0 MHz
Error: invalid response while writing the register 0x00 (count = 0, value = 7)

--- OR --- 
Error: invalid response while writing the register 0x00 (count = 0, value = 0)   

```
depending on whether the RST pin is hooked up or not.  value=0 corresponds to RST being hooked up.

ataradov, 2019-08-07:
Value 7 basically means "no response". The response code is is 3 bits and the default pulled-up value of the SWDIO is "1". 

Value 0 is an invalid value. I'm not sure under what conditions it can actually happen. I don't think simply shorting SWDIO to ground will do that. Or may be it will.

In any case this indicates some hardware issue.

ataradov, 2019-08-07:
This can also happen if the core is not properly powered for example. Can you share the schematic around the MCU (connection of all power pins)?

closed 2021-05-16 by ataradov