ataradov/free-dap #11: Difference between samd11 and samd11_vcp platform?

ooxi, 2020-01-01:
I tried looking at the sources and it seems like the vcp platform is using a more modular approach while also implementing cbc (serial console?) next to hid.

Is samd11_vcp intended to replace the samd11 platform? If so, why keep the old source around?

ataradov, 2020-01-01:
VCP version is newer and it does indeed implement a COM-port in addition to HID. This required a better approach to USB. So as a "reference" USB stack going forward I would use that one.

I do not intend to make VCP version hardware available for sale. I made a few standalone boards to test and I'm currently actively using one. But the main goal for this design was to be integrated into breakout boards like this one for example https://github.com/ataradov/breakout-boards/tree/master/stm32g071c / https://github.com/ataradov/breakout-boards/blob/master/pictures/stm32g071c_02.jpg

Having integrated VCP here is convenient. I'm not sure it adds value on a standalone debugger.

In theory it would be a good idea to port VCP version of the stack to the non-VCP hardware, but in reality it does not matter. The one that is there works fine.

ooxi, 2020-01-01:
I guess VCP = virtual console port, thus you have an integrated usb to serial converter for your breakout boards and do not need an additional ftdi chip?

Sounds nice!

closed 2020-01-01 by ooxi

ataradov, 2020-01-01:
VCP is a "Virtual COM-port". This is the standard USB name for this thing.

Well, I would never have FTDI chip, but yes, the idea is that I get a serial port to the target for free assuming I wanted to put D11 for programming anyway.