ataradov/free-dap #13: Reading from target on raising or falling clock edge?

ooxi, 2020-01-06:
According to [silabs](https://www.silabs.com/documents/public/application-notes/an0062.pdf), the target will both sample and put data on the line on a rising clock edge.

However to my understanding free-dap reads from the target on the falling clock edge e.g. [dap_swd_read_fast](https://github.com/ataradov/free-dap/blob/master/dap.c#L256):

```
    DAP_CONFIG_SWCLK_TCK_clr();
    bit = DAP_CONFIG_SWDIO_TMS_read();
    DAP_CONFIG_SWCLK_TCK_set();
```

Obviously free-dap works, thus I'm wondering whether my understanding of raising/falling edge or the documentation is wrong. Could you maybe point me in the right direction?

Moreover which documentation did you use when implementing SWD big banging?

ataradov, 2020-01-06:
The spec I used was debug_interface_v5_2_architecture_specification_IHI0031D.pdf 

But I don't think i gave much though to the clock polarity. I'm pretty sure I was just messing around and figured what works in practice.

IHI0031D says "When the target samples SWDIO, sampling is performed on the rising edge of SWCLK. When the target drives SWDIO, or stops driving it, signal changes are performed on the rising edge of SWCLK.

The default state of the SWCLK is high, so clock must go low first. I think it works because of the turnaround or idle cycles that provide the clock transition that actually makes the target prepare the data, which is samples after the negative clock transition in the read function.

ooxi, 2020-01-07:
Thanks for the information! I'm going to play around with SWD some more and come back if I need additional clarification.

closed 2020-01-07 by ooxi