ataradov/free-dap #15: No way to determine response size

JakeSays, 2021-03-07:
`dap_process_request` doesn't return the number of bytes written to `resp`, which forces the underlying transport to always transmit full size packets.

ataradov, 2021-03-07:
This is not a bug. USB HID only works if you transmit full size packets as described in the HID descriptor. OSes reject shorter frames.

JakeSays, 2021-03-07:
I'm actually using CANFD as the underlying transport, and having to transmit full size packets is really inefficient.

JakeSays, 2021-03-07:
Hmm. The response size is always encoded in the packet, isn't it. I'll use that.

ataradov, 2021-03-07:
Where do you see it encoded?  I don't think there is an expected response size anywhere.

It would not be too hard to add the actual response size, but I don't have time to do it now

JakeSays, 2021-03-07:
I misspoke. Actually its only encoded in the packets with variable length data.

ataradov, 2021-03-07:
And right now it is actually easy to overflow the buffer by requesting more reads than would fit into the response buffer.

So counting the bytes for the main data transfers is a good idea anyway. So I'll keep this issue open.

JakeSays, 2021-03-07:
Ok sounds good.

ataradov, 2021-03-07:
I've implemented buffer size checking and return of the response size.

The request buffer is still assumed to be a full packed size at maximum. There is no way to provide the actual size. But that should not matter that much.

ataradov, 2021-03-07:
And I've only tested that main functionality did not break. I do not know if the buffer sizes are reported correctly. Check it and see if it works.

closed 2021-03-07 by ataradov