ataradov/free-dap #21: Off by one error in dap_info()

newbrain, 2022-03-04:
A small off by one error in dap_info:
https://github.com/ataradov/free-dap/blob/0644042b940cc281c67cc1b48844d7e3d7045673/dap.c#L833

The correct value to use is `dap_resp_ptr - 2`, as the packet already contains the DAP_Info code (0x00) and the length.
E.g.: with the version string "1.3.0", `dap_resp_ptr` would be 1 (code) + 1 (length) + 5 (string) + 1 (appended 0) = 8 in line 833, the value we need is 6 (i.e. string length including final NUL).

ataradov, 2022-03-05:
Fixed.

closed 2022-03-05 by ataradov

newbrain, 2022-03-05:
It took you less time to fix and push it than for me to clone the repo!
Thanks!

ataradov, 2022-03-05:
I have not tested it, but it makes sense. 

I'm working on CMSIS-DAP v2 support and I'll do more testing once it is done. I've got it working on Linux and now I'm trying to figure out heaps of the microsoft BS you need implement just to have your device work on win.

newbrain, 2022-03-05:
I'm testing it with a pico as free-dap and a pico target.
Everything seems OK till now (flashing, breakpoints, etc.).

Thanks again, looking forward to the new release!