ataradov/free-dap #28: can't  flash other target with keil5 , keil5 not identify dap

songqiying, 2022-12-10:
[image]
Hi！ I try use this dap to flash other target with keil5. it not be identified correctly。 but the openocd can identify .  Is there something wrong with USB Device ？

ataradov, 2022-12-10:
What exact version do you use and what version of the free-dap did you use?
I just tried with Keil v5.36.0.0 and I built SAM D11 version with HAL_BOARD_VCP_V3. And it worked fine, Keil could detect the device.

songqiying, 2022-12-11:
very thanks for your feedback.  
the free_dap_rp2040.uf2 is my used firmware at  "bin"  directory , and the keil version is v5.23 .
I update the keil verison  to v5.36.   it work .  the problem is the keil version is to low.

closed 2022-12-11 by songqiying

ataradov, 2022-12-11:
Yes, I think older versions had issues with the HID descriptor format. It changed at some point and that change was necessary because new versions of the Mac OS had some issues enumerating the devices with the old descriptor. 

This is the commit that changed it https://github.com/ataradov/free-dap/commit/89b5247a93a4b242cbbbd1e6e00df129108b96e9 It was quite some time ago, and I guess Keil only recently got around to updating this stuff.