ataradov/free-dap #29: Cannot load free_dap_rp2040.elf via gdb

dlly11, 2023-05-22:
Problem:
When trying to load free_dap_rp2040.elf via GDB, GDB reports the file was written to the target but the target remains in BOOTSEL mode.

Tested loading picoprobe.elf onto the target and the target is flashed successfully.

Setup:

Debugger: RPi Pico running picoprobe
Target: RPi Pico

OpenOCD Config:
Version: 0.12.0

Config File:
```
source [find interface/cmsis-dap.cfg]
transport select swd
adapter speed 1000

set USE_CORE 0
source [find target/rp2040.cfg]
```

Steps to reproduce:

1. Compile free-dap firmware in platform/rp2040
2. Run `openocd -f <config listed above>`
3. Run `arm-none-eabi-gdb free-dap-rp2040.elf`

In GDB terminal

1. `target remote :3333`
2. `load`
3. `monitor reset init`
4. `continue`

GDB output:
```Reading symbols from free_dap_rp2040.elf...
(gdb) target remote :3333
Remote debugging using :3333
0x000020e0 in ?? ()
(gdb) load
Loading section .boot, size 0x100 lma 0x10000000
Loading section .text, size 0x2bdc lma 0x10000200
Loading section .data, size 0x24 lma 0x10002ddc
Start address 0x10000000, load size 11520
Transfer rate: 5 KB/sec, 3840 bytes/write.
(gdb) monitor reset init
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
(gdb) continue
Continuing. 
```

`lsusb` output
```
Bus 000 Device 008: ID 2e8a:000c 2e8a Picoprobe CMSIS-DAP  Serial: E66118604B644623
Bus 000 Device 010: ID 2e8a:0003 2e8a RP2 Boot  Serial: E0C9125B0D9B
```

ataradov, 2023-05-22:
Elf file does not contain CRC, which is required by the RP2040. You can either use bin2uf2 tool and add the CRC (it can generate raw binaries without making it  uf2 image) and program that separately from the debug session.

Alternatively you can do that once, note the CRC value, place it in the linker/rp2040.ld instead of the dummy value 0xcccccccc. You will have to do this again if contents of the boot section changes or changes to the compiler make it compile into other binary code.

dlly11, 2023-05-22:
Thanks! I'll try and add it using your tool.

dlly11, 2023-05-23:
After adding the CRC value manually to the linker the generated elf file was able to be flashed via GDB fine.

Thanks for your help

closed 2023-05-23 by dlly11