ataradov/free-dap #38: What is BOOT_ENTER?

nammenam, 2025-01-19:
I see that the pin A31 is set as BOOT_ENTER on the samd11 implementation, this happens to be the SWDIO pin as well.
In main, if this pin is set high a system reset triggers. 
Is this meant for flashing the debug unit and do i need it if I have connected up the nReset pin of the debug unit?

ataradov, 2025-01-19:
It is so you can enter the bootloader by simply shorting it to the ground. The bootloader has corresponding code where it stays in the BL mode if this pin is low on reset.

You can comment it out if it interferes with debugging and it is entirely useless if you are not running with a bootloader anyway.

nammenam, 2025-01-20:
Thank you so much😊

closed 2025-01-20 by nammenam