ataradov/free-dap #41: Unable to connect to STM32G030 using free-dap, but works with ST-LINK

james1236, 2025-03-05:
I have been attempting to connect and flash the 8 pin [STM32G030J6M6](https://www.st.com/resource/en/datasheet/stm32g030c6.pdf) chip using free-dap (via a Pico 1, using your pre-built [free_dap_rp2040.uf2](../tree/master/bin/free_dap_rp2040.uf2)) with no external components except for decoupling capacitors. The microcontroller was factory fresh and unflashed at the time of testing. I can see what appear to be typical SWD signals on an oscilloscope, but I can't connect to the target device. Replacing the free-dap Pico with an ST-LINK V2 allows me to connect perfectly fine.

### Trying with free-dap on pico (unsuccessful)

Signals while running openocd with free-dap on pico - SWDIO (green), SWCLK (blue) and nRST (purple/orange). nRST is driven low by the pico:

[image]
[image]
(the target chip is the middle one on the breakout board)

My openocd command is:
```openocd -f stm32g0x-custom.cfg -c "program data.hex verify reset exit"```

And my `stm32g0x-custom.cfg` file is:
```
source [find interface/cmsis-dap.cfg]
transport select swd
source [find target/stm32g0x.cfg]
adapter speed 480
reset_config srst_only srst_nogate connect_assert_srst
```

Openocd log (free-dap on pico):
```
C:\Users\James\Downloads\openocd-20240916\OpenOCD-20240916-0.12.0\bin>openocd -f stm32g0x-custom.cfg -c "program app.hex verify reset exit"
Open On-Chip Debugger 0.12.0 (2024-09-16) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 d52e355daa09f17ce64819122cb067b8a2ee0d4b
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
srst_only separate srst_nogate srst_open_drain connect_assert_srst
Info : Using CMSIS-DAPv2 interface with VID:PID=0x6666:0x9930, serial=1F9EA0D1
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 1F9EA0D1
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : Connecting under reset
Info : CMSIS-DAP: Interface ready
Info : clock speed 480 kHz
Error: Error connecting DP: cannot read IDR
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked
```

The log after running the openocd command is identical whether anything is connected to the pico or not, as in I can unplug the pico from the breadboard and nothing changes as far as openocd is concerned. This log is the same with all the openOCD versions and forks I tested on windows and macos.

I have also tried connecting to it with your edbg.exe:
```
> edbg.exe -t stm32g0 -c 100 -pv -f data.hex --verbose
Debugger: Alex Taradov Generic CMSIS-DAP Adapter 1F9EA0D1 2.0.0 (SJ, 12)
Using CMSIS-DAP v2
Clock frequency: 100.0 kHz
Error: invalid response during transfer (count = 0/1, status = 5)
```

### Trying with ST-LINK v2 (successful)

If I take out the pico, and put an ST-LINK-v2 in its place with SWDIO, SDCLK, nRST, GND and VREF connected, along with a 3.3V power supply, then I am able to connect and flash using openocd

[image]

When using the ST-LINK-v2 with openocd, I leave the command the same but change `stm32g0x-custom.cfg` to use stlink as below:
```
source [find interface/stlink.cfg]
transport select hla_swd
source [find target/stm32g0x.cfg]
adapter speed 480
reset_config srst_only srst_nogate connect_assert_srst
```

The consistently successful flash when using stlink:
```
C:\Users\James\Downloads\openocd-20240916\OpenOCD-20240916-0.12.0\bin>openocd -f stm32g0x-custom.cfg -c "program app.hex verify reset exit"
Open On-Chip Debugger 0.12.0 (2024-09-16) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 d52e355daa09f17ce64819122cb067b8a2ee0d4b
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_assert_srst
Info : clock speed 480 kHz
Info : STLINK V2J45S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.171317
Info : [stm32g0x.cpu] Cortex-M0+ r0p1 processor detected
Info : [stm32g0x.cpu] target has 4 breakpoints, 2 watchpoints
Info : [stm32g0x.cpu] Examination succeed
Info : [stm32g0x.cpu] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32g0x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x08007020 msp: 0x20002000
** Programming Started **
Info : device idcode = 0x10016466 (STM32G03x/G04xx - Rev Z : 0x1001)
Info : RDP level 0 (0xAA)
Info : flash size = 32 KiB
Info : flash mode : single-bank
Info : Flash write discontinued at 0x08003148, next section at 0x08007000
Warn : Adding extra erase range, 0x08003148 .. 0x080037ff
Warn : Adding extra erase range, 0x08007ce8 .. 0x08007fff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked
```

This instructables writer encountered a similar issue: https://www.instructables.com/CMSIS-DAP-V2-Programmer-Simple-With-Raspberry-Pi-P/

I'm not sure what I could be doing wrong, so I'd really appreciate your help, thanks!

ataradov, 2025-03-05:
I will have a look at that later today. But just for a test, try to lower the clock a lot (to like 10 kHz) with edbg. "status = 5" is likely due to timing issues, possibly caused by the breadboard setup.

I know that free-dap and edbg in general work with STM32G030, but I think I only tried on the SAM D11 platform.

james1236, 2025-03-05:
Thank you for your quick reply!
Here's lots of different speeds tested:

[image]

ataradov, 2025-03-05:
Yeah, something is broken here, for sure. Hopefully I can reproduce it on my end.

Also, as a side note, edbg only accepts raw binaries, so it will program the hex, but it will just put the hex into the device. This has no impact on the connectivity issue, of course.

ataradov, 2025-03-05:
I have STM32G031 as a target, which is soldered to a simple breakout board and connected with flying wires to the Pi Pico. First test results - works fine at 700 kHz and fails with status = 5 at 800 kHz:

```
alex:/tmp$ edbg -b -t stm32g0 -pv -f test.bin -c 700
Debugger: Alex Taradov Generic CMSIS-DAP Adapter A48AFD3E 2.0.0 (SJ, 12)
Using CMSIS-DAP v2
Clock frequency: 700.0 kHz
Target: STM32G031/41
Programming................... done.
Verification................... done.

alex:/tmp$ edbg -b -t stm32g0 -pv -f test.bin -c 800
Debugger: Alex Taradov Generic CMSIS-DAP Adapter A48AFD3E 2.0.0 (SJ, 12)
Using CMSIS-DAP v2
Clock frequency: 800.0 kHz
Target: STM32G031/41
Programming...Error: invalid response during transfer (count = 5/12, status = 5)
```

 I don't know where this is coming from. It might be long wiring, or the device itself. 5/12 suggests that the issue happens in the middle of a long transfer, so it is not just an outright fail. Something is marginal.

But it certainly should work at slower clocks.

In your case, it is hard to tell from the picture, but it looks like you are using VSYS(39) pin for the power. This pin will have ~5 V when powered from the USB. This will put STM32 outside of its operating range. Double check that.

ataradov, 2025-03-05:
Yeah, it looks like it is sensitive to timing and poor signal conditioning. I replaced intentionally log (8")  wires with short (3") and it works pretty much at max clock:

```
alex:/tmp$ edbg -b -t stm32g0 -pv -f test2.bin -c 8000
Debugger: Alex Taradov Generic CMSIS-DAP Adapter A48AFD3E 2.0.0 (SJ, 12)
Using CMSIS-DAP v2
Clock frequency: 8.0 MHz
Target: STM32G031/41
Programming................... done.
Verification................... done.
```

So, in your case I would verify voltages, since it should work as slower clocks.

james1236, 2025-03-05:
Thank you, sorry, I haven't used a pico in a while so I assumed VSYS was 3.3V and VBUS was 5V. I'll try tomorrow but hopefully that should fix it!

closed 2025-03-06 by james1236